JohnySeven/TWatchSK #86: No sound is played on notification received

miggat, 2022-02-11:
When a new notification is received it is properly displayed on screen and the device vibrates, but there is no sound at all. I sticked it to my ear to make sure it wasn't an audition problem on my side.

Here's what I can see on the serial monitor:

```
I (567777) WS: WEBSOCKET_EVENT_DATA
I (567777) WS: Received opcode=1
I (567787) WS: Got SK notification notifications.environment.wind.speedApparent with state emergency, active=0
I (567787) WS: Got message Delta from websocket with len=350
W (567797) WS: Received={"context":"vessels.urn:mrn:imo:mmsi:224372630","updates":[{"$source":"simple-notifications","timestamp":"2022-02-11T10:12:55.254Z","values":[{"path":"notifications.environment.wind.speedApparent","value":{"state":"emergency","method":["visual","sound"],"timestamp":"2022-02-11T10:12:55.253Z","message":"The Massa vent is more than 10.00 (m/s)"}}]}]}
W (567827) WS: Total payload length=350, data_len=350, current payload offset=0

I (568367) GUI: Show message 1073638088, event=0, message code=0!
I (568377) ASYNC: Starting task vibrate
I (568377) I2S: Playing sound alert repeat=3
I (568377) GUI: pending_messages_ empty, so msg added: The Massa vent is more than 10.00 (m/s), 11:12 am
I (568387) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32
I (568397) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (568407) I2S: I2S initialized OK!
I (568417) GUI: There are 1 messages in pending_messages
I (568417) GUI: Displaying the first message in pending_messages: The Massa vent is more than 10.00 (m/s)
I (568977) ASYNC: Task finished vibrate
I (571527) I2S: Sound playing alert finished
```

Let me know if I can provide further information to diagnoze the issue.

miggat, 2022-02-14:
I'm afraid i'm not skilled enough to test that example (TTGO_TWatch_Library fgails to build because some xtensa related issues that i don't understand) but i tested the factory firmware (https://github.com/sharandac/My-TTGO-Watch) and when i trigger an alarm, sound is properly played, and definetly loud enough.

Uploaded TWatchSK again, and still, same output as the original post, but no audible sound.

JohnySeven, 2022-02-14:
Fair enough, if other firmware works fine on your watch I'll need too look into this! Thanks!

miggat, 2022-02-15:
Just FYI, I manage to build and install the suggested sample (https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/BasicUnit/PlayMP3FromPROGMEM) and confirmed sound is playing.