JohnySeven/TWatchSK #87: Display Brightness doesn't show up

PatrikOls, 2022-02-15:
Display view doesn't show the "Display Brightness". Using a V3.0 Watch

ba58smith, 2022-02-15:
Can you take a picture of what that screen looks like, and post it here, please?

PatrikOls, 2022-02-16:
This is how the screen looks like. 
[image]

ba58smith, 2022-02-16:
Ah, yes. Now that I see it, I remember - I removed that setting with https://github.com/JohnySeven/TWatchSK/pull/80. It was no longer useful, as you can read in the PR description.

PatrikOls, 2022-02-16:
I can't get my watch to toggle any brightness. Don't know what brightness level I have, but it's to dark and I can't make it brighter.

https://user-images.githubusercontent.com/99754864/154278017-aa9f78ff-45a6-4808-b60d-fc4dd841464d.MOV

JohnySeven, 2022-02-16:
You've metioned you have V3 watch, right? The code is tested and works fine on V1, let me see if there is something I can do.

JohnySeven, 2022-02-17:
There are different Pins for running backlight on each version unfortunatelly. Will have to make V1-V3 firmwares, that would take few days and also TWatchSKDesigner will need to be updated to allow user to select correct version.

PatrikOls, 2022-02-17:
Fantastic if that could be done! We are a long time cruisers and I love my SignalK server, extend it to a Twatch is something that will be very useful. Our yacht Sigrid af Skane at anchor.

[image]

JohnySeven, 2022-02-17:
Nice, will hopefully have something this weekend done.

PatrikOls, 2022-02-22:
Did a quick fix and changed the .h files in  .pio\libdeps\ttgo-t-watch\TTGO TWatch Library\src\board

Changed the:  TWATCH_TFT_BL              (gpio_num_t)12                                to 
                       TWATCH_TFT_BL              (gpio_num_t)12PIN 15 

Works on my V3 now.

JohnySeven, 2022-02-22:
@PatrikOls I've finally managed to add support for V2 and even V3 - as I don't have hardware yet I can't test the Firmware on real device. May I ask you to download recent version of TWatchSK Designer for your OS: at https://github.com/JohnySeven/TWatchSKDesigner/releases/tag/v1.1 and test the current firmware uploaded via designer?

Note: during firmware upload please don't check "Clear device flash prior upload" as it would erase SPIFSS and your settings would get lost.

Thanks,
Jan

JohnySeven, 2022-02-22:
@PatrikOls also PR #88 contains compilation support for V3 variant, feel free to try compiling the code and even respond to the PR with any issues or success!

PatrikOls, 2022-02-27:
Have now tested using my TWatch V3and it works great! I used the TWatchSk Designer on Windows 11 with success.
Have also compiled and upload the firmware  using Windows 11 with success. 

Had to lower the upload speed to get the firmware to be uploaded when using Windows.

[image]

Many thanks!!!!

JohnySeven, 2022-02-27:
Great, glad it works, please feel free to post any issues or new feature requests if you have one.

As it all works fine, I'm closing this one.

Thanks!

closed 2022-02-27 by JohnySeven