JohnySeven/TWatchSK #93: K to F Temperature Conversion Incorrect Ver 1.2

RichFind, 2022-05-22:
K to F
With an input of 281.25 the conversion produces an output of 176.62 when the result should be 46.58

I don't think that °F = (K − 273.15) × 1.8 + 32 can be easily implemented in the current scheme.

My own fault for living in a country that refuses to accept the world standard.

JohnySeven, 2022-05-23:
@RichFind good point, we'll look into it ASAP.

ba58smith, 2022-05-23:
I think you just need to rework the equation so that it's one multiplication and one addition. I'll look into it later today.

ba58smith, 2022-05-23:
@RichFind, use this formula, which you can implement in TWatchSKDesigner.
Fahrenheit = Kelvin × 1.8 - 459.67
https://www.rapidtables.com/convert/temperature/how-kelvin-to-fahrenheit.html

closed 2022-05-23 by ba58smith