JohnySeven/TWatchSK #97: Labels are not displayed

miggat, 2023-04-10:
I'm not sure if this project is abandoned, but I wanted to use my Twatch again this week, formatted it and added new views using the Twach SK Designer 1.2.

After clicking on "Download DynamicViews" and watiting for the restart, the views are there, but I can only see a the button and the switch.

[image]

However, the 2 labels doesn't appear on the watch. The space they should take is there though, but it's completely empty,

JohnySeven, 2023-04-10:
Could you share your json file? Labels are working, maybe there is some issue with settings of those labels :-) I see you're C# guy, am I correct?

miggat, 2023-04-10:
Yup, I work mainly with C#, don't know how you knew that :O

Here's the json for this test screen:

`{
  "views": [
    {
      "name": "New view",
      "layout": "column_mid",
      "components": [
        {
          "text": "TWS",
          "font": "montserrat14",
          "type": "label",
          "size": [
            100,
            40
          ]
        },
        {
          "text": "--",
          "font": "montserrat28",
          "binding": {
            "path": "environment.wind.speedOverGround",
            "multiply": 1.944,
            "offset": 0.0,
            "decimals": 0,
            "period": 1000
          },
          "type": "label",
          "size": [
            100,
            40
          ]
        },
        {
          "text": "test",
          "font": "montserrat14",
          "put": {
            "path": "test.number",
            "value": 1
          },
          "type": "button",
          "size": [
            100,
            40
          ]
        },
        {
          "type": "switch",
          "size": [
            60,
            40
          ]
        }
      ],
      "type": "normal",
      "background": "white",
      "size": [
        0,
        0
      ]
    }
  ]
}`

I must add than when I first tested this (at that time there were only labels, no gauge, button or switch), it used to work perfectly fine.

Oh, and also I can see data is being received on the console:

```
(141462) WS: WEBSOCKET_EVENT_DATA
(141462) WS: Received opcode=1
(141462) WS: Got SK value update environment.wind.speedOverGround
(141472) WS: Got message Delta from websocket with len=207
(141472) WS: Total payload length=207, data_len=207, current payload offset=0
```

JohnySeven, 2023-04-10:
> Yup, I work mainly with C#, don't know how you knew that :O

Yeah, because you have Xamarin Forms fork you are mobile developer! Non C# guy would fork Xamarin Forms!

Also checkout my new - hot project https://github.com/JohnySeven/SKipper it's KIP replacement I'm working on done in C# and Avalonia UI - runs everywhere!

miggat, 2023-04-10:
> Yeah, because you have Xamarin Forms fork you are mobile developer! Non C# guy wouldn't fork Xamarin Forms!
> 
> Also checkout my new - hot project https://github.com/JohnySeven/SKipper it's KIP replacement I'm working on done in C# and Avalonia UI - runs everywhere!

haha yes, that was from a veeery old project :P 

I made a quick look to SKipper and liking it so far. I put it a star so I follow it's evolution.

JohnySeven, 2023-04-10:
> I made a quick look to SKipper and liking it so far. I put it a star so I follow it's evolution.
Thanks! Also checkout screenshots on my Mastodon account: https://social.dytrych.cloud/@jan/tagged/Skipper they will give you better look than readme and source code :-)