JohnySeven/TWatchSK #99: Support for new S3

JohnySeven, 2023-09-13:
https://www.aliexpress.com/item/1005005642999869.html?pdp_npi=3%40dis%21USD%2154.98%2147.83%21%21%21%21%21%402116417d16890469527112042d0416%2112000033861406916%21affd%21%21&spm=a2g0o.detail.1000023.5.6b1ay9EFy9EFTU&aff_fcid=21b94a4d86714d5db21cf9d77e8aa42d-1694599945731-04881-_DBCyDN7&tt=CPS_NORMAL&aff_fsk=_DBCyDN7&aff_platform=portals-billboard&sk=_DBCyDN7&aff_trace_key=21b94a4d86714d5db21cf9d77e8aa42d-1694599945731-04881-_DBCyDN7&terminal_id=804cf05314fe43fa9fcdf7685bf16f9e&afSmartRedirect=y

miggat, 2024-01-07:
I just received a new S3, What kind of help do you need?

miggat, 2024-01-11:
Are you planning to add support for S3? I made my own attempts but i'm not skilled enough I'm afraid. And it qould be really cool to have it working

JohnySeven, 2024-01-11:
Well I would like to add support for S3, but I don't have any time frame when I will be able to actually do it. Without Hardware it will be very difficult to do that, because the TICKLESS Arduino that I'm using might not support ESP32-S3 so I'll have to make a work around or find Arduino version that supports it.